polyvision/EasyRaceLapTimer #1: Unable to download distribution image

PProvost, 2015-11-20:
I tried more than four failed attempts to download the dmg.zip file from your server and the file either failed checksum verification or simply didn't download at all. Perhaps hosting it somewhere else would help? Or opening anonymous FTP access? I suspect it is your web server getting stuck at the end of the file.

PProvost, 2015-11-20:
Cancel that. The checksum was fine on my most recent attempt, now the issue is figuring out what to do with it. The usbit tool you mention doesn't seem to know about .dmg.zip files, and the file won't unzip using 7-zip on windows.

Suggestions?

polyvision, 2015-11-20:
Hey there. I'm uploading a new image at present which is just 8GB big. This should fit on any SD card >= 8GB. Realized yesterday that 32GB are not the same for each manufacturer of SD cards...

closed 2015-11-20 by polyvision

peisel, 2015-11-20:
Can i use this with WIndows?

PProvost, 2015-11-20:
Thanks @polyvision. Can you post when the new image is available? Also, why is it a .dmg file instead of a .img or imz file?

peisel, 2015-11-20:
http://www.easyracelaptimer.com/downloads/ 0.2 is online