polyvision/EasyRaceLapTimer #9: Race Restart Button didn't work?!

peisel, 2015-11-29:
Update via commands - not the complete image.

I use a buzzer button - contact closed as long i pressed the button. Connected to 22,23 pin. Did i miss something?
[image]

peisel, 2015-11-29:
Sorry i see it in my own picture ... wire wrong ... ;-)

peisel, 2015-11-29:
Sorry wiring is ok - issue exist

polyvision, 2015-11-29:
same here, try again after the restart ;)

peisel, 2015-11-29:
hmm ... how should it work ... i thought the current race will reset? or how should this button work?

polyvision, 2015-11-30:
if u push the button, a new race will get started

peisel, 2015-12-02:
Ich bekomme es nicht zum laufen ... der schalter verbindet pin 23 und gnd solange wie ich ihn gedrückt halte. Ich hatte einmal einen Eintrag "Race November 29, 2015 22:40	0	0	0.0s		0.0s" dies konnte ich aber nicht reproduzieren, Ich benutze img 0.2 mit dem manuellen update auf 0.3. Den Pin habe ich auch einmal mit Kabel auf gnd überbrückt aber dort das gleiche liegt also nicht am schalter.... ratlos

polyvision, 2015-12-03:
Ich habe jetzt gerade nicht das Layout vor mir. Aber Du hast + an den PIN-Eingang gelegt? Hm, aber warte bis ich nachher nachschauen kann, bevor ich jetzt Mist rede ;)

peisel, 2015-12-03:
Der Taster verbindet das Signal von PIN 23 (Rotes Kabel) mit GND Pin 25 (schwarzes Kabel) wenn er gedrückt wird, ich gehe davon aus etwas spinnt mit software oder pi nicht?

polyvision, 2015-12-04:
wie lange drückst du denn? also 1sek muss gedrückt werden damit etwas passiert, ist so in der software programmiert

peisel, 2015-12-05:
Keine Ahnung ... lange gedrückt, kurz gedrückt ... ich warte mal ab auf 0.4 ...

peisel, 2015-12-07:
when i connect pin 23 and 24 for a second it works ... is it right? Pin 23 reacts when its connected with a + signal?

closed 2015-12-08 by peisel