polyvision/EasyRaceLapTimer #14: Setting an ID Without Reprogramming

frezik, 2015-12-07:
I noticed that the racer's unique ID has to be set in the code, with a maximum of 63 IDs.  It would be nice to be able to set this ID prior to a race without a full flash of the chip.

There are a few ways this could work. I2C, SPI, and Serial should all be available on the ATtiny85 chip. SPI in slave mode is a little wonky on Arduino (the library doesn't support slave mode directly). All three may have timing issues in the Arduino library due to the lack of a 16MHz crystal driving the chip.

A crude but effective means would be similar to how memory chips work. You set binary data with one bit per wire (across however many wires you have), and then "latch" a signal on another wire. When the chip sees the latch happen, it reads the values off the data wires to build the word.

With the IR LED taking up one wire, the ATtiny85 only gives us 3 data wires plus 1 latch. By doing this twice, you could send 6 bits, for 64 possible values.

A slightly easier way is to bump up to an ATtiny2313 and use a whole 8 wires. Comparing prices on Digi Key, this would add less than $1 to the cost for single orders, and an almost insignificant cost on bulk orders. The pin headers for this are also a tiny cost.

A further extension could be to use the Ardunio EEPROM library:

https://www.arduino.cc/en/Reference/EEPROM

Which would allow for maintaining the ID in between reboots.

polyvision, 2015-12-09:
There will be a new revision of the  transponders in the future. But at present, I'm focussing on the software.

polyvision, 2016-01-11:
Solved in a new rev of the transponders. Soon to be released.

closed 2016-01-11 by polyvision