polyvision/EasyRaceLapTimer #19: Update Documentation

peisel, 2015-12-08:
Please add commands for:
- clean up race history
- clean up racers
- Reset User Passwords
- edit "Admin" Account

polyvision, 2015-12-09:
What do you mean by "clean up racers" and "clean up race history"? One single button to delete them all?

peisel, 2015-12-09:
Not a ring to rule them all ... a button to delete all --- better would be a list functionality for delete ... like select all and unselect some ... to delete most of them ...

PProvost, 2015-12-17:
If you want to toss the entire database and start clean, you can do this:

    RAILS_ENV=production rake db:drop db:create db:migrate db:seed

The first command (drop) tosses the current database, then create, migrate and seed recreate it (similar to the steps in the setup guide).