polyvision/EasyRaceLapTimer #22: Monitor API - update speed

FieserKiller, 2015-12-13:
Hi, 
I built an app which announces lap times, check it out: https://play.google.com/store/apps/details?id=de.my_fuzzy_logic.laptimeannouncer
Tested it on field today. We had 3 smartphones polling the interface every second. It worked, but I noticed a varying delay between a copter passing the sensor and his last_lap time changing in monitor. 
It is usually ~5sec behind, sometimes a few seconds more or less.

polyvision, 2015-12-13:
Hi,

yes that's true at present. I'm firing and forget the "send lap time" requests at present, QT handles then the requests.
I'm planning to change this portion of the code, so that these requests will arrive faster in the rails app.

FieserKiller, 2015-12-13:
The perfect solution from the point of view of my app would be broadcasting monitor JSON over UDP to the network, whenever data changes. 
This way we would cover 3 topics at the same time:
- Service discovery. Every device would find EasyLapTimers address easily by listening to UDP broadcasts on its WiFi interface
- No polling anymore, less load, less bandwith and power usage
- (Almost) real time monitoring data for everyone :)

polyvision, 2015-12-21:
It's implemented now, please test it ;)
Port is 33333.

There's now also a config value udp_broadcast_address which defaults to: 192.168.42.1

Maybe 127.0.0.1 is a better solution, dunno. First time, I'm doing something with UDP broadcasts.

polyvision, 2016-01-04:
The update speed in v0.4RC1 should be now even faster

closed 2016-01-04 by polyvision

FieserKiller, 2016-01-04:
Back from holidays - will test and implement into LaptimeAnnouncer App :) Thx

fape, 2016-11-06:
Do you know why udp broadcast was removed on 0.6 branch? see 222493ca67ae545545f2e10878e69f29285fb60d