polyvision/EasyRaceLapTimer #24: lap_timeout_in_seconds don't work (b0.4)

ChrisVonFrosta, 2015-12-13:
on my last system (o.2 updated to 0.3) the lap_timeout_in_seconds didn't work. 
I seem to remember that peisel was saying the same.

Now with a fresh new image (b0.4 ) it'S semms to be still the same.
[image]

polyvision, 2015-12-13:
I will have a look.
Uhm... just to clearify. lap_timeout_in_seconds  means, that only all XX seconds, requests are getting accepted for a new lap time

peisel, 2015-12-13:
Set lap_timeout_in_seconds to 10 seconds ... but in monitor i have laps under 10 seconds ... this shouldn't be possible ...

ChrisVonFrosta, 2015-12-13:
You could see in the screenshot that the fastet lap are less 4 seconds. my set timeout time is the default 4 seconds

polyvision, 2015-12-18:
changed the config values in v0.4b2, now there's also lap_min_lap_time_in_seconds.

closed 2015-12-18 by polyvision

ChrisVonFrosta, 2015-12-18:
how could i upgrade my v0.4b too v0.4b2 ? is it possible?

polyvision, 2015-12-18:
Yes,

it's not official released yet, but you can do it this way:

cd ~/EasyRaceLapTimer/
git pull origin master
cd ~/EasyRaceLapTimer/web
bundle
RAILS_ENV=production rake db:migrate
RAILS_ENV=production rake assets:precompile