polyvision/EasyRaceLapTimer #27: Building Transponder

sniperxnl, 2015-12-16:
I see no crystal on the Attiny85? Is that for a reason?

sniperxnl, 2015-12-16:
I have build the same on

1x Attiny85
1x 330Ohm
1x 56Ohm
1x NPN BC547
1x Vishay 950NM (i wait for the osram)

So i think i build a next one with crystal 8 MHZ and 2 x 10 pF on the gnd.

What do you think?

PProvost, 2015-12-17:
What do you think you need a crystal for? Are you thinking you need an external oscillator?

The transponder code expects the ATTiny to be running at 8MHz which the built-in oscillator is quite capable of handling. Trust me. I have a dozen of these working on my desk using nothing more than an ATTiny85, a 2n222A, two resistors and the IR LED.

polyvision, 2015-12-17:
Hey, there. You don't need a crystal. Like PProvost said, the internal oscillator works just fine in 8Mhz mode. Just burn the bootloader with 8Mhz (like described in the docs) and you're done.

closed 2015-12-21 by polyvision