polyvision/EasyRaceLapTimer #37: Feature Request: Multicolor LED output for gates based on last pilot through.

c3h5, 2016-02-16:
I typically fly in small groups all via FPV and basically we cannot keep track of each other.
So I had an idea to allow color assignment to pilot’s transponder IDs in the software and have a hardware output from the Raspberry Pi to set multi colored LED.
The purpose would be so that the Raspberry Pi could control the multicolored LED display attached to a gate so that the pilots flying could see who flew though a gate last.
Example my friends color is blue and I see the gate ahead is blue thus he beat me to it.
But then some laps later I see a gate ahead and it’s my color, thus I know I’ve lapped him.

Hope the above makes sense

I think this would be great in a multi gate setup
and also having the option of the gates auto clearing of color after a users predefined amount of time.

frezik, 2016-02-16:
I happen to have been working with RGB LEDs of late. An easy way to do this is with GPIO, which would give you the 3 primary RGB colors, plus the 3 secondaries by combining any two of the 3 primaries.

It's possible to get any arbitrary color using PWM on each of the RGB colors, but the Pi only has 1 PWM output. You can use an Arduino to take an i2c or serial signal that sets the color, and then use its own PWM outputs to the LEDs.

c3h5, 2016-02-17:
This would be fine as the number of people flying at the same time is typically small and we care more about a quick head to head battle than say a best time. And powering a Pi/LED display is no big deal when you have plenty of old 3s lipo nobody is using. :)