polyvision/EasyRaceLapTimer #40: Add Compatibility for Ilap Transponder and Clean Flight

mattyra, 2016-02-27:
Hello...

So there are number of Transponders coming out at the mo... but personally i like the open source but at the club some have different transponders... 

Is its possible to add a Ilap and Clean Flight Compatibility to the system ? if so what do you need 

Matty

SirVodkat, 2016-04-11:
This would be a great addition, seconded!
Is this useful: http://blog.owenson.me/reversing-ilap-race-transponders/

frezik, 2016-04-11:
This is certainly possible. We need a specification (perhaps a reverse engineered one) to be able to implement it.

Right now, the v2 daemon can handle the protocols for both the 0.3 and 0.4 transponder protocols, but not both at the same time. I think I have a way to handle multiple protocols in the same run.

I was told that i-lap has a particularly complicated and fast protocol, so it may not be possible.

Differences in the carrier wave frequency may also require different receiver hardware.

polyvision, 2016-04-13:
Hey there,

I think we won't need it. Just wait a few more days for an announcement by myself. Maybe I've found a really cool and easy to way to track lap times. It's not a new solution but a simple one for pilots.

slacker87, 2016-08-26:
I plan on merging easyracelaptimer support into my already ilap compatible transponders in the near future, so that will always be an option for those that want to have both! Polyvision may I reach out for integration q's if I have any?

github.com/slacker87/coreir