polyvision/EasyRaceLapTimer #43: IR Protocol Examples

frezik, 2016-03-02:
I'm not quite sure if the docs are incorrect or if I'm missing something, but I think the examples have a space that shouldn't be there:

```
id 14: PULSE 300 | SPACE 300 | SPACE 300 | SPACE 300 | PULSE 600 | SPACE 600 | PULSE 600 | SPACE 300 | PULSE 300
id  3: PULSE 300 | SPACE 300 | SPACE 300 | SPACE 300 | PULSE 300 | SPACE 300 | PULSE 600 | SPACE 600 | PULSE 600
```
My understanding is that every section should alternate between pulses and spaces, so the above should be:

```
id 14: PULSE 300 | SPACE 300 | PULSE 300 | SPACE 300 | PULSE 600 | SPACE 600 | PULSE 600 | SPACE 300 | PULSE 300
id  3: PULSE 300 | SPACE 300 | PULSE 300 | SPACE 300 | PULSE 300 | SPACE 300 | PULSE 600 | SPACE 600 | PULSE 600
```

tb942, 2016-05-04:
ive made a pull request for this bug

tb942, 2016-05-05:
This has now been merged and fixed, can you close the issue?