polyvision/EasyRaceLapTimer #86: Powering the transponder

tlayh, 2016-08-22:
Hi, I am currently trying to build the transponders for the EasyRaceLapTimer in version 0.2. My question now, is it possible to power the transponder with 6v instead of 5v? I was thinking about adding a small battery pack for powering the transponder. That way I could use it without soldering some wires to the quad. Best way for a battery pack I found are to use two cr2032 with a switch to power it on and off.

Greetings,
Thomas

frezik, 2016-08-22:
The attiny85 datasheet lists 5.5V as its max operating voltage. You might get lucky, but I wouldn't count on it.

You could get away with powering at 3V, but the LED will be dimmer since the current-limiting resistor is higher than it needs to be at that voltage.

Any reason you don't want to take power off the flight controller or something like that? The Naze32 (and similar FCs) almost always have a spare 5V output pin somewhere.

tlayh, 2016-08-22:
Thanks for your answer. Main reason for not using the FC for me would be to allow guest pilots to easily add the transponder. But I think it should work if I put two diodes in front, they should lose 1V and then I can use the 6V battery pack. I let you know if it works ;)