polyvision/EasyRaceLapTimer #94: Unable to fetch files

stevelee10, 2017-05-27:
Hi hope you can help out, i have been following the setup guide, but its having some errors where it is unable to fetch some of the files from mirrordirector.raspian.org

is there a particular reason for this?

[image]

pandalion98, 2017-05-28:
Sometimes it's just down.

stevelee10, 2017-05-28:
ill try it again tonight after work, just thought i would check i hadnt done anything stupid.

pandalion98, 2017-05-28:
@stevelee10 If it's still down or if you want additional reliability, pick some mirrors from [https://www.raspbian.org/RaspbianMirrors](https://www.raspbian.org/RaspbianMirrors). It won't hurt to add a few more. Just follow the file format and you'll be fine.

After that, do an `apt-get update` and you should be good to go.

stevelee10, 2017-05-28:
cheers, its still not playing, so ill give that ago, its only the 1 file it fails to fetch, bit annoying
cant access that site either i get a 504 gateway timeout

stevelee10, 2017-05-29:
ok ive found another working mirror, how do i change mirror so that it looks at this one instead?

stevelee10, 2017-05-29:
ive tried other mirrors im having loads of issues, my latest is it cant locate package libaprutill-dev

stevelee10, 2017-05-30:
sorted now i think. thanks for the help

closed 2017-05-30 by stevelee10